Ultimaker/Cura #16353: About function of Help causes program to be unresponsive

bpybas, 2023-08-01:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Prusa Mk3

### Reproduction steps

1. Start program
2. Select About from the Help menu
3. Program is unresponsive and does not display the About window
4.  According to the Task Manager Cura is running 3 components 1. Console Windows Host; 2. Slicing Engine; 3. UltimakerCura

### Actual results

Ultimaker Cura 5.4.0 is unresponsive.

### Expected results

The About window should open and give all the appropriate details.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-08-03:
Thanks for the report.
I'm on Windows 10 also.  Because I might have 3 or 4 versions of Cura open at a time I use the "About" command a lot (can't remember which is which you see) and I haven't noticed any problem with it.
Could you leave some information about the computer system you are running into the problem on (manufacturer, CPU, GPU, etc.).  The Cura team will need more information.

bpybas, 2023-08-03:
Here is a screen print of the Windows 10 System About page.
[image]

bpybas, 2023-08-03:
Further info as provided by the PrusaSlicer 2.6.0 System Info

WorkingSet: 209MB
PrivateBytes: 364MB
Pagefile(peak): 364MB(837MB)
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 4.6.0
Profile: Core
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
GLSL version: 4.60.0

bpybas, 2023-08-07:
I did the About function in CURA 2.6.0 again today, and let it just sit for a couple of hours.  Still no response from the running app, so I went in through the Task Manager and Ended the CURA task.  It came up with a message of Slicer Failed to Complete.