Ultimaker/Cura #16355: Elegoo Neptun 4/Neptun 4 Pro

jakabandras, 2023-08-02:
### Is your feature request related to a problem?

Please add printers Elegoo Neptun 4 series. Thank you

### Describe the solution you'd like

I can use my new printer.....

### Describe alternatives you've considered

I can't use my new printer with Cura and must to search alternative slicing software....

### Affected users and/or printers

Eleggo Neptun 4series

Roxxor91, 2023-08-29:
Hello!
I would like to participate in adding support. I own a Neptune 4 (!NOT! Pro) and imported the extruders and printer definitions from elegoo cura into cura 5.4. It is my first 3D printer, but i get consistent perfect results, so I guess it is fine.
Here are the files:
[Neptune 4.zip](https://github.com/Ultimaker/Cura/files/12469356/Neptune.4.zip)

I tested the Neptune 4 defintion as I own this printer. Now I am new to this project. How are we going to add this configuration?
Why does the config use 2 extruders? The devices obviously just have one.
I would appreciate help on how about to go forward.

Roxxor91, 2023-08-29:
@Ev1dentFir3
Can you have a look? Your commit for the 4 pro is missing a lot of detailed speed settings, isnt it? Or am I missing something?

xin5683, 2023-08-30:
This is the official technical support repo from ELEGOO, providing 4pro support. I have verified that it can be used and hope it can be helpful to you.
[https://github.com/NARUTOfzr/Cura-Neptune-TEST](https://github.com/NARUTOfzr/Cura-Neptune-TEST)

bmullen90, 2023-08-31:
I would also appreciate it being added to the latest version. I'd love to have all of my printers there in one program instead of a bunch of different ones.  Thanks!

Ev1dentFir3, 2023-08-31:
> @Ev1dentFir3 Can you have a look? Your commit for the 4 pro is missing a lot of detailed speed settings, isnt it? Or am I missing something?

You mean the ones in the inherited "elegoo_base" definition?

mastercaution, 2023-12-09:
Can be closed due to #16819 (merged)

GregValiant, 2025-12-09:
I'll go ahead and close this.
Elegoo no longer sends printer definitions to UltiMaker as they have their own slicer.

closed 2025-12-09 by GregValiant