Ultimaker/Cura #16359: Body and arm

Leontudo, 2023-08-02:
### Cura Version

5.2.2

### Operating System

Windows 10

### Printer

Ender 3 s1

### Name abnormal settings

Model crashing wiht default "Standard Quality" profile

### Describe model location

If I move it it slices correctly

### Describe your model

Yes i sliced it before but closed and forgot to save the file in the usb, after that I wasn't able to slice anymore

### Add your .zip here ⬇️

[Failing project.zip](https://github.com/Ultimaker/Cura/files/12235353/Failing.project.zip)

GregValiant, 2023-08-02:
Thanks for the report and the file.
Try moving the model around on the build plate.  You might also try rotating the model or changing the Initial Layer Height by a little bit.  I didn't have any trouble slicing it, but there is a bug that keeps good models from slicing in certain orientations.

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic