Ultimaker/Cura #16363: crash slicing with creality ender 3 S1

thealchimiste, 2023-08-02:
### Cura Version

5.4.0

### Operating System

WINDOWS 11

### Printer

ender 3 S1

### Name abnormal settings

Yes but it crashad with default profiles

### Describe model location

no with rotate and move

### Describe your model

never succeded, yes watertight, mesh fix done

### Add your .zip here ⬇️

[CE3S1_paris_Imported_FINAL.zip](https://github.com/Ultimaker/Cura/files/12238901/CE3S1_paris_Imported_FINAL.zip)

GregValiant, 2023-08-02:
Thanks for the report and the file.
There aren't any errors in the model.  I used all my tricks and I can't get it to slice either.
I'll mark this as a duplicate and the Cura team will take a look.

thealchimiste, 2023-08-02:
Thanks for your confirmation.
I take care to verify the file in details before to send it.
Have a nice day.

GregValiant, 2023-08-02:
I decided to check the model again.  I used [formware.co/OnLineSTLRepair](https://formware.co/OnlineStlRepair) and it came back with "No Errors" but it did add a bunch of vertices and triangle.  When I brought the repaired file in to Cura it sliced with my Ender 3 Pro and my setting profile, but not with your S1.  It looks like a bug to me.

GregValiant, 2023-08-02:
4.13.1
[image]

thealchimiste, 2023-08-02:
May I to understand that with this version 4.13.1 it works ?

GregValiant, 2023-08-02:
Yes.  4.13.1 is the last version that did not include the Arachne Engine.  It's very stable and works well.

thealchimiste, 2023-08-02:
It works to slice ! I enter my printer parameter.
In all case I add to reduce to 94 %.
It seems to calculate space of surface differently. It consider a grey zone...
I don't know if I can change paramater to put out the grey zone to avoid to reduce...
Thanks for the solution in all case  !
[image]

GregValiant, 2023-08-02:
The gray zone is where the skirt/brim will fit.  If you turn off "Bed Adhesion" the gray zone should go away.  If you change the "Skirt Distance" or the number of loops then the gray area will change size.

Cura 5.4 introduced a different way to do that so the gray zone never appears.

thealchimiste, 2023-08-02:
It works !! Great.
I have to work on profil now but I keep this version in case of problem with the new engine.
Thanks for your help.

closed 2024-04-03 by HellAholic