Ultimaker/Cura #16364: top layers are partly missing

Pipo1024, 2023-08-02:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Toolchanger from duet 3D

### Reproduction steps

just slice export and print

### Actual results

some other level toplayers missing

### Expected results

normal closed surface

### Add your .zip and screenshots here ⬇️

[toplayer missing.pdf](https://github.com/Ultimaker/Cura/files/12239028/toplayer.missing.pdf)
[CFFFP_Body6.zip](https://github.com/Ultimaker/Cura/files/12239053/CFFFP_Body6.zip)

GregValiant, 2023-08-02:
In the "Top/Bottom" settings you have "Top Layers" set to "0".  Click the "fx" icon next to the setting to link it back to your profile.
Let us know if that works for you.
I'll go ahead and remove the bug label and close this.  I hope you understand.
[image]

closed 2023-08-02 by GregValiant

Pipo1024, 2023-08-03:
Sorry for reporting.

Stupid mistake from my side

 

 <https://user-images.githubusercontent.com/64202104/257790109-fa203e81-85ed-4941-8162-4f3380615891.png> 

Dit e-mailbericht is met AVG-antivirussoftware gecontroleerd op virussen.
www.avg.com