Ultimaker/Cura #16366: Slicing Failed But File is slicing on 5.3.1

mikeywaldron, 2023-08-02:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 v2

### Describe your model

i have used mesh tools and model errors while attempting to slice.

### Add your .zip here ⬇️

[Great_Lion_Wall_Light_0d33f4a0ca.zip](https://github.com/Ultimaker/Cura/files/12244316/Great_Lion_Wall_Light_0d33f4a0ca.zip)

GregValiant, 2023-08-03:
Thanks for the report.
Upon opening the file this warning came up.
[image]

From formware.co/onlinestlrepair
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 3 Non-manifold edges (?)
--> 4 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 53692ms ------
-> Vertex count changed from 31235 to 51689 (**+20454**)
-> Triangle count changed from 68020 to 121976 (**+53956**)

Here is the repaired model.
[image]

It still didn't slice until I moved it away from the midpoint of the build plate.  That's a known bug so I'll mark this as a duplicate and the Cura team will take a look.

mikeywaldron, 2023-08-03:
i have just tried all of that and wont slice at all :) slices fine without repair using 5.3.1 but not 5.4

GregValiant, 2023-08-03:
That might be a clue to the bug.  I've seen the "Print thin walls" option giving trouble in other reports.
There is no problem slicing this in 4.13.1 either.

mikeywaldron, 2023-08-03:
Yeh it fully sliced on 5.4 now without repair. Also one of the models you've said couldn't slice the freighter i got to slice now. Print think walls seems to of bugged it out a bit as soon as i disabled that everything is slicing again