Ultimaker/Cura #16367: Extrusion speed limit

mhier, 2023-08-02:
### Is your feature request related to a problem?

When printing very soft materials, the speed usually has to be reduced quite drastically to prevent the material from escaping between extruder and hotend. The printing speed currently has to be defined in X/Y printer movement speed, so the maximum possible speed for these materials depends on the layer height and extrusion width. Whenever changing one of those, the speed settings have to be adjusted as well.

### Describe the solution you'd like

I propose to introduce a new setting which limits the maximum extrusion speed (flow) in mm/s directly. This limit is in effect in addition to the defined printing speeds in X/Y, so that the lower speed wins.

### Describe alternatives you've considered

I don't see alternatives except keeping it as is and configuring this manually every time.

### Affected users and/or printers

Everyone printing materials that need to be printer specially slow.

### Additional information & file uploads

This will be especially useful when combining with the Material Settings plugin (https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin), since the material-dependent extrusion speed could be placed into the material profile while the X/Y printing speed can be configured in the quality profile.

GregValiant, 2023-08-03:
This is a duplicate of several other requests for some sort of "Feed Rate control by extrusion rate".  #16284 is an example of one but a search around here will turn up others.

mhier, 2023-08-03:
All right, I am sorry, this did not appear during my searches (it is hard to find something across 2.6k issues if you don't exactly know what to look for). I will add my 2ct to #16284 :-)

closed 2023-08-03 by mhier