Ultimaker/Cura #16368: Slicing failed

dusanfpwn97, 2023-08-02:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Pro

### Describe your model

Slicing fails when slicing with 100% scale. With 95% it works fine

### Add your .zip here ⬇️

[handring.zip](https://github.com/Ultimaker/Cura/files/12244464/handring.zip)

GregValiant, 2023-08-03:
Thanks for the report.
Note that the model has errors.  That always makes it tough to figure out if Cura has a bug or if the model is the problem.
[image]

This is a report on the model errors.  The "Non-manifold" edges are a particular problem as they confuse the slicer regarding where the exterior of the model ends and the interior starts.
--> 40 Naked edges (?)
--> 2 Planar holes (?)
--> 0 Non-planar holes (?)
--> **768 Non-manifold edges (?)**
--> 775 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 15118ms ------
-> Vertex count changed from 75000 to 76312 (+1312)
-> Triangle count changed from 151450 to 154478 (+3028)

After repairing the model it sliced.
[image]

Cura 5.4 does have a bug that can keep models from slicing, but in this case it looks like the model is the problem.
I'm going to remove the bug label from this report and close it.  I hope you understand.

closed 2023-08-03 by GregValiant

dusanfpwn97, 2023-08-03:
Of course. I was just confused why it worked with <= 95% scale and below, but not with 100%. Cheers!

GregValiant, 2023-08-03:
Scaling down can "close" open seams while scaling up can cause seams that seemed closed to open up.  I know it was only a 5% difference, but computers are working to 12 or more decimal places.  That 5% may have pushed it within some slicing tolerance.