Ultimaker/Cura #16371: slicing failed

kyloren1970, 2023-08-03:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

anycubic vyper

### Describe your model

its a black panther helmet and it sometimes slices but most of the time the slicing fails

### Add your .zip here ⬇️

[AV_CWBlackPantherHelmetSplitFrontV2.zip](https://github.com/Ultimaker/Cura/files/12245756/AV_CWBlackPantherHelmetSplitFrontV2.zip)

GregValiant, 2023-08-03:
Thanks for the report.
This was a tough one to get to slice.  I'll mark this as a duplicate as this behavior has been noted many times.
After moving the model and rotating the model I changed the Initial Layer Height to 0.23 and it sliced.  When I turned on Support, it quit slicing.
The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic