Ultimaker/Cura #16372: slice doesnt work

Warbud666, 2023-08-03:
### Cura Version

5.2.2

### Operating System

windows 11

### Printer

kobra

### Describe your model

freightliner cascadia

### Add your .zip here ⬇️

[CFFFP_Freightliner Cascadia 2011 - Solido - 0-08-r.zip](https://github.com/Ultimaker/Cura/files/12246754/CFFFP_Freightliner.Cascadia.2011.-.Solido.-.0-08-r.zip)

GregValiant, 2023-08-03:
Thanks for the report.
I don't find any errors in the model and it just won't slice.
I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic