Ultimaker/Cura #16373: Slicing issue

alfonsvh, 2023-08-03:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Name abnormal settings

See project file

### Describe your model

Sliced before without issue

### Add your .zip here ⬇️

[UMS5_Disso-Cup_LID-withprintedseptum.zip](https://github.com/Ultimaker/Cura/files/12249257/UMS5_Disso-Cup_LID-withprintedseptum.zip)

GregValiant, 2023-08-03:
Thanks for the report.
I moved the parts away from the midpoint and it sliced.  This is a known bug so I'll mark this as a duplicate.

On the small inner part - every other layer has this going on.
[image]

After using the "Lay Flat" tool every other layer looks like this with a small concentric circle inside.
[image]

Did you intend for the small part to be skewed?

alfonsvh, 2023-08-03:
No, that isn't supposed to happen...

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic