Ultimaker/Cura #16374: Slicing failed on large base with figures contained inside the base

IanVogt, 2023-08-03:
### Cura Version

5.4.0

### Operating System

Windows 11 Home 22H2

### Printer

Creality Ender-3

### Name abnormal settings

Fails with Tree and Normal support. Does not fail in version 4.12.1 with Normal support.

### Describe model location

Only fits on the build plate centered and unrotated.

### Describe your model

Two chess pieces on a board.

### Add your .zip here ⬇️

[white2playH.zip](https://github.com/Ultimaker/Cura/files/12251529/white2playH.zip)

GregValiant, 2023-08-03:
Thanks for the report and the file.
Well, it sliced for me on Win10 with no problem but that has happened a lot lately.  I don't know if it is system specific, or a regular ole bug, but there seem to be a lot of "won't slice" reports for 5.4.0.
I'll mark this as a duplicate and the Cura team will look.
[image]

IanVogt, 2023-08-04:
I don't know if this helps but on my machine (Windows 11 22H2) version 5.4.0 will slice with scaling at 52.5% but not 53%.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic