Ultimaker/Cura #16375: Missing some geometry after slicing

KillCoffe, 2023-08-03:
### Cura Version

5.4

### Operating System

MAC OS Ventura 13.5

### Printer

FLSUN Super Racer

### Describe your model

cylinder

### Add your .zip here ⬇️

[BUG.3mf.zip](https://github.com/Ultimaker/Cura/files/12254583/BUG.3mf.zip)

It's a downloaded model in which I have "attached" a small cylinder in order to have an easier surface..

Attached you have what it should like and how it looks like when I slice it.. I have used SuperSlicer and OrcaSlicer but none of them had the same problem, they managed the stl correctly.. only cura is having problem, I've also tried to "fix" the stl but no results..

let me know if you need further informations. 
<img width="800" alt="Screenshot 2023-08-03 alle 20 24 45" src="https://github.com/Ultimaker/Cura/assets/59095395/8a46eda9-b76b-4d1b-a58f-01b9
[image]
cc92fecf">

GregValiant, 2023-08-03:
I get an error when I try to open the bug.3mf.zip file.  You will need to try again.

KillCoffe, 2023-08-04:
[bug.zip](https://github.com/Ultimaker/Cura/files/12257330/bug.zip)

please, try with this one

GregValiant, 2023-08-04:
That worked.

1. Your Printer Definition File has overrides that set the "Gantry Height" to "0.0" and "machine_head_with_fans_polygon" to [0, 0, 0, 0].  The result of them all being set to 0 is that you cannot print multiple parts in "One at a Time" mode.  A problem with this project is that you **_have_** the Print Sequence set to "One at a Time".  Change that to "All at Once".
2. It is nice to have a "Setting Profile" so you can grab it without having to change a lot of settings every time you load a model.  The downside is that you tend not to "look" at the settings to see if they are actually appropriate.  Your "Hole Horizontal Expansion is set to 0.1.  For this particular model - set "Hole Horizontal Expansion" to 0.0.  Since you are using a brim which attaches to the base of the model, you should also set "Initial Layer Horizontal Expansion" to 0.0 as well.

The model on the left is with your settings and the model on the right has "Per Model" Hole Horizontal Expansion set to 0.0.  (The Print Sequence is set to "All at Once" or this would not slice because "One at a Time" is illegal with your printer definition overrides and UM Cura is not responsible to 3rd party definition files.)
[image]

So given your settings (and the Printer Definition overrides) what I see is "expected behavior" rather than a bug in Cura.  I'll remove the bug label and close this.  Let me know if you feel that is premature.

closed 2023-08-04 by GregValiant