Ultimaker/Cura #16379: Updates to wiki "Running-Cura-from-Source" page

nickzoic, 2023-08-04:
### Cura Version

5.5.0-alpha

### Operating System

Linux

### Printer

N/A

### Reproduction steps

Building from source on Linux based on https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

> External developers must run `conan remote remove cura`. Internal developers can contact IT about getting Artifactory Access.

This  doesn't seem to be the case ... it seems like I've got access to your artifactory from out here? ... so that's either a documentation issue or a security issue, your call :-)  An earlier version of this document[*] also mentions `use -r cura-ce instead of -r cura if you're a community developer`  ... if there's a remote external developers *should* be using please let us know!

[*] https://github.com/Ultimaker/Cura/wiki/Building-Cura-from-Source/7f61da9954f5d78fbd1206ea2681a21b76245940 which I think means its no longer in the wiki but Google still finds it?

### Actual results

it works!  So I've got a 5.5.0-alpha build here which is nice.

### Expected results

n/a

### Add your .zip and screenshots here ⬇️

n/a

MariMakes, 2023-08-14:
Hey @nickzoic,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This issue is very much outside my field of expertise 😦😖
But I'll bring it up to the other developers to see what we can do to improve it.

jellespijker, 2023-09-23:
Hi @nickzoic it is intended that everybody has access to that Artifactory, otherwise it would be damn hard for community developers to build Cura. We switched to another private Artifactory repository for any proprietary and building steps.

I'm closing this ticket as it the Wiki in the repo is up-to-date and we have limit control over the search results in the cache of Google

closed 2023-09-23 by jellespijker