Ultimaker/Cura #16391: Slicing Failed-Merged Models

Lobo-Mostla, 2023-08-04:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.5

### Printer

Ultimaker S5

### Describe your model

a model for dual printing

### Add your .zip here ⬇️

[UMS5_octopus_part2_origin.3mf.zip](https://github.com/Ultimaker/Cura/files/12263688/UMS5_octopus_part2_origin.3mf.zip)
[image]

GregValiant, 2023-08-05:
Thanks for the report and the file.
It wouldn't slice for me either.  I changed the Initial Layer Height to 0.27 and it sliced.

I'll mark this as a duplicate report and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic