Ultimaker/Cura #16396: Slicing fails if I rotate the model

mpodroid, 2023-08-05:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1 (c)

### Printer

Creality Ender-3 S1 Pro

### Describe model location

Slicing works if model is vertical. If horizontal it fails

### Describe your model

base platform for my car dashboard, to host phone, GPS and other accessories mounts

### Add your .zip here ⬇️

[models.zip](https://github.com/Ultimaker/Cura/files/12267015/models.zip)

GregValiant, 2023-08-06:
Thanks for the report and the files.
It would not slice for me until I rotated the model 90°.  No matter what I tried it would not slice the two models at the same time.

I'll mark this as a duplicate for the Cura team.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

mpodroid, 2024-04-05:
All. Thanks. Well done.