Ultimaker/Cura #16398: Tree support is missing in Cure 5.4

KoronGIT, 2023-08-05:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Elegoo Neptun 3 Pro

### Reproduction steps

Slicing a model in Cure 5.4 with tree support. 

### Actual results

Tree support is missing in required places

### Expected results

The tree support is created correctly as in 5.3.

### Add your .zip and screenshots here ⬇️

I did the whole thing once with the exact same settings in Cure 5.3 and Cura 5.4. Also reset to default settings and also move, rotate and change the layer height of the model.

The problem persists.

In Cure 5.3 it is correct: [image]

At Cura 5.4. is it wrong. I once marked it with green circles.
[image]

ThomasRahm, 2023-08-05:
Set `minimum support area` to 0 and it should work.

KoronGIT, 2023-08-05:
> Set `minimum support area` to 0 and it should work.

This change worked and the support is now generated correctly.
[image]

GregValiant, 2023-08-06:
Thank you @ThomasRahm .
I'll go ahead and close this.

closed 2023-08-06 by GregValiant