Ultimaker/Cura #16402: [5.4.0]"Drop Down Support Mesh" in "Per Model Settings" not visible upon ticking box

Jerdeen, 2023-08-05:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Biqu BX / Ender 5 Pro

### Reproduction steps

1. add desired model for slicing and custom support model
2. select your custom support model
3. go to per model settings>print as support>select settings>show all>scroll to bottom>enable drop down support mesh

(This method was functional in versions before Cura 5.4.0)

### Actual results

ticking "drop down support mesh" yields no changes

### Expected results

"drop down support mesh" should be visible enabling the user to disable the setting for supports with overhangs.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

in previous versions the setting was visible like so: (taken from google images)
[image]

cventin, 2023-08-27:
I am having the same issue. Selecting to show any of the 'per model' settings does not result in the settings being made visible. This problem persists even if the project is saved, Cura restarted, and then the project reopened.

daniels220, 2023-12-12:
I am also having this problem on Cura 5.6, macOS Ventura. Super annoying, makes creating custom support structures almost useless/impossible as they cannot have any overhangs.

@DrWho198 Not sure what you mean—what "per model settings icon" are you talking about other than the one in Jerdeen's first screenshot?