Ultimaker/Cura #16406: Wrong diameter in calculation

maxpd1, 2023-08-06:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Anycubic Mega 

### Reproduction steps

Loaded same STL File as in Cura 4.13 into fresh installation of 5.4. 
Tested different files.
Sliced project with default material and printer settings: 1.75 filament diameter, 0.4 nozzle, draft 0.3mm
Cura 5: 2h41m, 55g, 6,9m
Cura 4.13: 2h40m, 54g, 18,22m

So I think internal calculation of cura 5 takes another diameter of filament as shown.

The final print isn't usable due to of a lack of filament used. 

### Actual results

Cura 5: 2h41m, 55g, 6,9m

### Expected results

Cura 5: 2h41m, 55g, 18m

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[kreiselstaender.zip](https://github.com/Ultimaker/Cura/files/12270208/kreiselstaender.zip)

GregValiant, 2023-08-06:
Thanks for the report.
I loaded your model in 4.13.1 (below left) and in 5.4.0 (below right).  Both versions had new installs of the i3 Mega.  You can see that the results are really close.
[image]

Load the model into Cura and use the "File | Save Project" command and zip the 3mf file and post the zip folder here.  I haven't run across this problem myself and this is the first report regarding this behavior.

maxpd1, 2023-08-07:
[AI3M_kreiselstaender.zip](https://github.com/Ultimaker/Cura/files/12284200/AI3M_kreiselstaender.zip)

Thank you for your investigation!

GregValiant, 2023-08-07:
I opened and sliced the project in 4.13.1 (on the left) and 5.4.0 (on the right) and the results are the same.
[image]

But look here.  The "Extruder" tab of Machine Settings shows 1.75 diameter while the "Printer Settings" plugin Diameter box is showing 2.85 diameter filament.  That box is showing the "circle with arrow" icon next to it indicating it has become unlinked from the Extruder box.  Clicking the circle arrow icon will straighten that out by re-establishing the link between the boxes.

See if that makes a difference for you.  The Printer Settings plugin is available in the MarketPlace.

fieldOfView, 2023-08-08:
Looking at the project file, you have somehow selected the 2.85mm variant of the PLA material, instead of the 1.75mm variant. This should not have been possible.

maxpd1, 2023-08-08:
I newly selected the only Generic PLA Material available. There is no 2.85 version of Generic PLA. But now calculation is correct.
But still strange why 2.85 were selected. I still never found anywhere this information (I didn't installed Printer Settings Plugin yet)
And still strange how this should have been changed as it was a out of the box installation without touching any settings from my end.

GregValiant, 2023-08-08:
I have seen problems with the Nozzle size boxes.  In my estimation they don't link back and forth correctly (Machine Settings to Printer Settings) and that same thing might be happening with the Material Diameter.  The problem seems to have started with 5.0 which debuted new dialog boxes.  Most of the problems were fixed for 5.1 and 5.2 but this one may have been unknown until now.

@MariMakes of the Cura team may take note and bring it up with the others.

MariMakes, 2023-08-09:
Hey All 👋 

What an interesting bug 🤔 
Thanks for reporting it 👍 

What I'm seeing is that when I load your project file the diameter is 2.85 in the printer settings, but when I select Generic PLA, it changes back to 1.75  That is indeed something that shouldn't happen. 🙃 
[image]

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞