Ultimaker/Cura #16410: CE3E3V2_NSCC_Logo_with_Base2

Psychofireman22, 2023-08-06:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ender 3 V2

### Describe your model

Never sliced

### Add your .zip here ⬇️

[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/12270783/New.Compressed.zipped.Folder.zip)

GregValiant, 2023-08-07:
Thanks for the report and the file.
It sliced for me with no changes.  You can try moving the model on the Cura buildplate, rotating the model, or changing the Initial Layer Height by a little bit.  Sometimes those can fool this bug.

I'll mark this as a duplicate since this has been reported numerous times.  The Cura team will take a look.

Psychofireman22, 2023-08-08:
Thank you, I will try that.
Michael C. McFalls

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

Psychofireman22, 2024-04-06:
Understood. I will look into it.
Thank you,

Michael C. McFalls