Ultimaker/Cura #16411: CURA 5.4.0 Failed slice

RaptyGB, 2023-08-07:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 Pro (Klipper)

### Name abnormal settings

N\A

### Describe model location

Fails to slice at any location\rotation

### Describe your model

Part of a Ship model

### Add your .zip here ⬇️

[Nostromo Part 6 117.zip](https://github.com/Ultimaker/Cura/files/12273973/Nostromo.Part.6.117.zip)

RaptyGB, 2023-08-07:
Got  new error after moving it at diagonal angles: error code 0x00007FFF946005964. not sure its relevant but more info is always good :-)

GregValiant, 2023-08-07:
Thanks for the report.
This "won't slice" bug can often be fooled in three ways...move, rotate, and/or change the "Initial Layer Height".  With the ILH at 0.23 this sliced for me.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

closed 2023-08-08 by RaptyGB