Ultimaker/Cura #16412: Strange extrusions in 'Taco Bell Bag' ("Vase" Mode)

Ergonomicmike, 2023-08-07:
### Cura Version

5.4.0

### Operating System

Win10

### Printer

Ender 3 v2, mriscoc firmware

### Reproduction steps

Download [the Taco Bell Bag from printables.](https://www.printables.com/model/159780-taco-bell-bag-vase-mode/files)

Set Cura to Standard Quality Profile.  Then enable Spiralize Outer Contour, which automagically enables Smooth Spiralized Contours.

### Actual results

There is a strange extrusion inside the bag. (See screen shot.) It starts at layer 301, and continues in a few more layers.

When I change the layer height to 0.24, and the layer width to 0.6, there are a different, but similar, strange extrusions.

Remarkably, I don't see a break in the spiralized wall, which I would expect from this deviation.

### Expected results

There should be no strange extrusion. (A la Prusa slicer.)

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2023-08-07:
Thanks for the report.
Those are spots where the "Spiral" had to stop and make an adjustment _but_ because you are in spiralize the extrusion doesn't end and that's what you see there.  If you turn on the "Travel" lines you will see that those bad extrusions also have travel moves associated with them and if you turn on "Starts" you will see a white dot at one end of each of those extrusions.
[image]
This is showing the travels that those starts cause.
[image]

This seems to be a problem inherent in the 5.x versions of Cura.  At this time I think 4.13.1 is a better choice.  For myself, I use the SmartAvionics fork of Cura for Spiralize.  It's been nicely optimized and is based on the 4.x version.  (It might not run on a MAC but works on Windows.)
This behavior has received a lot of discussion both here and on the UM Forum.
The Cura team will take a look.

Ergonomicmike, 2023-10-11:
This slice bug is still present in Cura 5.5.0-beta.1. Looks to be exactly the same as in Cura 5.4. See image where the model is centered at 0,0,0.

However, when I move the model enough, the slice bug goes away, per GregValiant. (See second screen shot.)

I do not know if moving the model hid the bug when I reported this in Cura 5.4.

[image]
[image]

Ergonomicmike, 2023-12-04:
@MariMakes Still the same slicing bug as in the top screenshot in Cura 5.6.0.

Ergonomicmike, 2024-03-30:
Issue is still unresolved in Cura 5.7 beta 1.

[image]

GregValiant, 2024-03-30:
When Cura 5.x slices this, there are areas that devolve into islands and that is where spiralize fails.
These are all images of the back left corner at layer 406.  (The settings are the same in all the slicers.)
This is 4.13.1.  This is what I would expect.
[image]

This is PrusaSlicer.  It is virtually the same as 4.13.1.
[image]

And then we get to Cura 5.x.  Here the path includes a small island that developed.  That island causes spiralize to fail and there is an extrusion across the print and a travel back.
[image]

The island doesn't have to be large.  Any island forces spiralize to consider it a separate piece on the particular layer.
This is a different area.  The "island" is just a dot, but forces a huge path change.
[image]

Ergonomicmike, 2024-04-02:
@GregValiant Glad that you're on it.

(Between you and me ("I" in Spanish), I've moved on to a "Whale" of a Slicer. Still, I'd like to see Cura succeed.)

GregValiant, 2024-04-02:
I play with PrusaSlicer, Orca, and Bambu.  I'm not a fan of the interface and post-processing is difficult.  The "customizability" of Cura is nice to have.
There have been numerous complaints about how Spiralize works in the 5.x versions of Cura.  I'm not discounting that, but that particular model has (what I consider to be) a problem at one of the corners.  Even the SmartAvionics fork of Cura (which is what I use for Spiralize) makes an island there and the spiral path is interrupted.
[image]

So it's an interesting model and I understand that the Slic3R based slicers get through it, but from what I've seen, too many of the "Scanned" models have issues.  For many of them, it's that they are ridiculously high resolution.  So much is dependent on the tech doing the scanning and how the point cloud is interpreted in whatever software they are using to translate the point cloud into surfaces.  It isn't easy, and they seem to often miss the mark.

Ergonomicmike, 2024-04-02:
Interestng about the problems with the model. I will run it thru OnLine Repair later to see if that fixes the problem.

I have become used to Orca-think. I admit that it doesn't have the Plug-In's that Cura has. (For thread support, which you steered me onto last year.)

GregValiant, 2024-04-02:
I don't see any "model problems" with the bag.  It's watertight and there aren't any duplicate or flipped surfaces.  There is a fold on the one edge and it is too narrow for Cura to get an "out and back" path into it.

This is the area that concerns me here.  There are a couple of pinched areas there.
This is in Orca and that area that shows the widely spaced extrusions is going to fail.  The extrusions through the pinched areas look like they will leave a raggedy edge, but the island doesn't form.
[image]

This is the same area in Cura 5.7b.  It develops into an island and so the spiral path fails.  I haven't been able to tune that island out of the slice.
[image]

The way I look at it is that the model just isn't right for Spiralize.  Along with the pinched area, there are also some areas that could really use support and support means spiralize quits because support is always an island on a layer.
Could Cura do a better job of slicing it?  I think so.  It used to.  Maybe the Arachne engine is getting in the way.

There are only so many software engineers that write code for apps like this.  I would be very surprised if the Cura team and the Prusa team weren't maintaining some kind of contact and talking about the issues they both face.

Ergonomicmike, 2024-04-03:
@GregValiant So should I close this Issue, since, from what you found, it's not truly an "issue" with Cura. Cura is doing the best it can with a flawed model. And GIGO. (Or feel free to close it yourself. Tnx.)