Ultimaker/Cura #16417: 5.4.0 Slicing failed -- with "Print thin walls" checked

csterlin, 2023-08-07:
### Cura Version

5.4.0

### Operating System

Windows 11 Home, 22H2, 22621.1778, Intel(R) Iris(R) Xe Graphics

### Printer

Ender 3

### Name abnormal settings

"Print thin walls" being checked caused the slicing failure in this case

### Describe model location

no

### Describe your model

Yes, with "Print thin walls" unchecked, yes, watertight, via 3dBuilder and Prusa Slicer, but not with meshtools which could not fix it.

### Add your .zip here ⬇️

The example is with the file scaled to 600% larger than the original.
[MChouse.zip](https://github.com/Ultimaker/Cura/files/12283744/MChouse.zip)
[5.4.zip](https://github.com/Ultimaker/Cura/files/12283752/5.4.zip)

Kiefels2, 2023-11-26:
I have recently ran into this problem AGAIN. Went away for a while but now it's back again, even on models it sliced the day previous. 
Tried the new beta 2 version and it does it in that also.
I have not added any new hardware or sofware, so at a loss on why this came back

closed 2024-01-18 by MariMakes

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0