Ultimaker/Cura #16419: 5.4.0 STL file cannot be inserted

AnyITil, 2023-08-07:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1

### Printer

Anycubic i3 Mega s

### Reproduction steps

Open an STL in Cura

### Actual results

It doesn't copy the STL in.

### Expected results

open the file after inserting it, to slice it

### Add your .zip and screenshots here ⬇️

****

MariMakes, 2023-08-08:
Hey @AnyITil,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We've received some reports with the shipped STL reader not working for everyone. 
Could it be that our STL reader plug-in is disabled by accident?
If you go to the marketplace, click on the gear, and scroll down until you see STL Reader, you should be able to Enable it again from there.
[image]

Can you let us know if that is your issue?
If that's not the case, just let us know we'll take another look. 👍
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/16249

AnyITil, 2023-08-21:
The STL reader isn't deactivated. I have observed that the files open, it takes a long time to load.

MariMakes, 2023-08-28:
Hey @AnyITil,

Do you see this happen with all models? 
Can you [try this model for example?](https://www.thingiverse.com/thing:1278865)
Could it be that you are trying to load a model with too many triangles or MBs?

If it's just a certain model than to me, it doesn't seem like a bug in Ultimaker Cura.

AnyITil, 2023-08-28:
Hey,

I tried it once and noticed that loading the Calibration Cube or other models took 4 minutes immediately after opening Cura. After the first model was loaded, I could then open other models in any MBs without waiting.

MariMakes, 2023-09-04:
Hey @AnyITil,

Thanks for the update. 
I'll bring it up to the team to see what they can do to improve it. 
We have some Mac developers, they might recognize the issue. 

Fingers crossed 🤞

MariMakes, 2023-09-11:
Hey @AnyITil,

I showed your issue to the team and they had a good idea how we might learn more about what is happening. 
To learn better about what's slowing your software down, could you install a plug-in that performs more extensive logging? 

You can download it here: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
After it's installed can you.... 
	1. Select Extensions
	2. Select Remote Support
	3. Start CPU profiler
	4. Repeat your steps to reproduce. 
	5. Stop CPU profiler
	6. Select Save Remote Support Archive
	7. Go to the Marketplace
	8. Select the gear icon 
	9. Uninstall Startup Optimiser
	10. Share the Remote Support Archive Zip file here on Github

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

AnyITil, 2023-09-11:
[remote-support_23-09-11-14_18_37.zip](https://github.com/Ultimaker/Cura/files/12575907/remote-support_23-09-11-14_18_37.zip)

rburema, 2023-09-15:
Thanks! However, from what I can see from those 'extensive logging', is that the STL loading takes about 49+ seconds, and spends most of that (48+ seconds) in the actual library code that's supposed to load the STL. Nothing that 'we' programmed seems to get in the way of that...

Was this file made when opening a file when you observed the behaviour, or after?

AnyITil, 2023-09-15:
After opening Cura, I started logging and opened a file. The error with the long loading time did not occur. Before the test, I deleted junk files from my Mac, which made it run significantly faster. I think this fixed the long loading time issue. Now I don't have any loading time problems. I still made the extensive logging, there could have been other errors.

rburema, 2023-09-15:
There don't seem to be other issues, not with the STL at least.

I think this issue can be closed.

closed 2023-09-15 by rburema