Ultimaker/Cura #16428: [5.4.0] creates infill blobs on top surface

laurvas, 2023-08-08:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender 3 with Klipper

### Reproduction steps

1. Open attached file in Cura 5.4.0, click "Center selected"
2. Take a default "Standard Quality" profile, change wall line count to 3, extra skin wall count to 0
3. Press Slice button and Preview

### Actual results

Note small infill blobs on the top surface of the wall in top left corner (see screenshot).
This bug don't reproduce on Cura 5.2.2, not tested on more recent versions.

### Expected results

No blobs

### Add your .zip and screenshots here ⬇️

[image]
[top surface bug.zip](https://github.com/Ultimaker/Cura/files/12292326/top.surface.bug.zip)

GregValiant, 2023-08-09:
Thanks for the report.
I'm not sure what I'm supposed to see here.  It looks normal to me.  You have "Starts" turned on for the preview and those are the white spots (indicating the Z seam) but I don't see any blobs and Cura wouldn't show them anyway.  Blobs are certainly something that can happen, but they aren't planned for.  You can get some blobs at the z seam and they can be made worse by using "Retract Before Outer Wall" since that will cause primes to occur at the Z seam.
This is with Starts turned off.  I just don't see anything "bad" here.
[image]

If you post a good photo of a print that shows what you are talking about it could be helpful.

laurvas, 2023-08-09:
I've marked that bad things with blue circles. There is no issue with Z seam.
[image]
Yes, I have "Origin at Center", that is my own legacy firmware customization. Don't think it makes any sense actually.