Ultimaker/Cura #16431: Staggered Layer Heights for Inner Walls

BradenMeyers, 2023-08-08:
### Is your feature request related to a problem?

The lack of waterproof capability of 3D prints is mainly from the seams at the layer changes from what I have observed in experiments. There are a lot of settings that can be adjusted to optimize waterproofing, but I imagine this setting would greatly increase the ability of a print to be water tight. 

### Describe the solution you'd like

Have an option to stagger the height of the walls. For example if you wanted to print 4 staggered walls the printer with a layer height of 0.4mm. After printing the bottom layer, the printer would go up 0.2mm and print the 2nd and 4th walls with a layer height of 0.2 then the printer would move up 0.2mm and print the 1st and 3rd walls with a layer height of 0.4mm then it would print the rest of the layer with infill. The printer would then move up 0.2mm and print the 2nd and 4th walls with a layer height of 0.4mm. Continue the pattern until you get to the top layers. The last wall layer could be a 0.2mm layer height for the 2nd and 4th walls. 

### Describe alternatives you've considered

I have tried almost all the settings in the slicer and tested to see what settings make the most waterproof prints. The main factors are perimeter walls and flow rate of top/bottom and wall flow.

### Affected users and/or printers

All users and printers.

GregValiant, 2023-08-10:
You want something done in the vertical which would require a lot of re-coding (read that as "money").  Applying Horizontal Expansion of 1/2 line width to every other layer might give you something to work with.  It wouldn't work with spiralize and would still depend on bonding though.

I paint the inside.

jaggzh, 2024-04-20:
I discussed this for years; but my idea of staggered heights was more for intended for strength than water tightness. I get water-tight prints with our normal printing.  Interlocking (staggered) layers/paths would help though.
A couple tips:
1. Shorter layers cause more compression (against the natural desire of the filament to round itself off. Ie. the "die swell"), but this can cause more "pushing" of one layer against the next.
2. ABS and acetone smoothing (or other materials and their solvents, like PVB+alcohol.. if that's suitable for you.  I usually want alcohol-resistance).  I am doing acetone smoothing almost every print (due to the projects I work on's needs).

Anyway, here's my old video from a couple years ago showing staggered layers. I might have posted it in some feature request before too:
https://www.youtube.com/watch?v=qARgOYFDlHI

CNC Kitchen did a video on it recently (apparently people are sending him requests for it).