Ultimaker/Cura #16432: wrong plate measures on ender3 profile

a-dom88, 2023-08-08:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender 3 (BASIC MODEL)

### Reproduction steps

1. I updated from an old version of cura (like 5.2 or even prior version) to the 5.4.0
2. I printed a bunch of stuff but then i realized that the dimension of the profile for standard (basic) ender3 are wrong
3. I found that the dimension in the default profile is 235x235 while the base ender 3 is 220x220. The new update has the plate measures of the ender3 v2, not the basic one.

### Actual results

I don't know if this is the right place to write about this problem but i'll just repeat what i did in the reproduction steps

1. I updated from an old version of cura (like 5.2 or even prior version) to the 5.4.0
2. I printed a bunch of stuff but then i realized that the dimension of the profile for standard (basic) ender3 are wrong
3. I found that the dimension in the default profile is 235x235 while the base ender 3 is 220x220. The new update has the plate measures of the ender3 v2, not the basic one.

### Expected results

I don't know if this is the right place to write about this problem but i'll just repeat what i did in the reproduction steps

1. I updated from an old version of cura (like 5.2 or even prior version) to the 5.4.0
2. I printed a bunch of stuff but then i realized that the dimension of the profile for standard (basic) ender3 are wrong
3. I found that the dimension in the default profile is 235x235 while the base ender 3 is 220x220. The new update has the plate measures of the ender3 v2, not the basic one.

### Add your .zip and screenshots here ⬇️

I don't know if this is the right place to write about this problem but i'll just repeat what i did in the reproduction steps

1. I updated from an old version of cura (like 5.2 or even prior version) to the 5.4.0
2. I printed a bunch of stuff but then i realized that the dimension of the profile for standard (basic) ender3 are wrong
3. I found that the dimension in the default profile is 235x235 while the base ender 3 is 220x220. The new update has the plate measures of the ender3 v2, not the basic one.

GregValiant, 2023-08-09:
Thanks for the report.
The definition files come from the manufacturer or from members of the community, and requests to alter them can come from the community.  In this case it's likely that some well-meaning person put in a pull request and altered the build plate size.
So this isn't an UltiMaker Cura problem as UltiMaker has no way to check the suitability of the definition files for the over 300 printers from other manufacturers.  I'll remove the bug label and close this.  I hope you understand.

I have an Ender 3 Pro and the glass measures 235 x 235.  Not all of that is printable.  The edges have a 1mm bevel all the way around so subtract that and you have 233 x 233.  The centerline of the nozzle has to stay a minimum of 1/2 line width from the edge so round that up and knock off another 1mm and you are at 232 x 232.  Lastly you need at least some tolerance for beds that might not be located perfectly since they "float" on a spring suspension.  Call it 1mm all around and you are at 230 x 230.  That is about the maximum "Printable Area" for the standard Ender 3 and 3 Pro.  The original definition files did have the bed listed at 220 x 220 in Cura which was extremely conservative as it left a much larger safety border.  That also drew a lot of complaints since the bed does measure 235 x 235.  People didn't like "losing" that 15mm (even though it really isn't a full 15mm).
This will give you a 230 x 230 printable area and your prints will be centered:
- Make sure there is no plastic sticking out of the nozzle.
- Auto-home the printer.
- Move the Z up 2mm
- Move the nozzle to a point 2.5mm in from the left edge of the build surface and 2.5 mm in from the front edge of the build surface.
- Move the Z back down to "0".
- On the LCD select "Set Home Offsets".
- In the Cura Machine Settings make the build plate size 230 x 230.

closed 2023-08-09 by GregValiant

a-dom88, 2023-08-09:
Hi, thank you for the response but the 220x220 is a limit that the nozzle can't exceed and you can see it because when you move the nozzle via control panel, it cannot go above 220 x or 220 y, so even if the plate is more than that, you cannot reach it, at least, i cannot make it reach it with the base model without any mods.

thank you for your support

GregValiant, 2023-08-09:
If your firmware is limiting the movement to 220 x 220 then that is an example of why the definition files cannot be checked for suitability by UltiMaker.  They have to trust the people submitting the definitions and pull requests.

My Ender 3 Pro definition file listed the bed size at 220 x 220 but it has 235 x 235 movement.  I was able to make the changes noted above and the "printable area" is indeed 230 x 230.  The print that is running now is using all of the width.