Ultimaker/Cura #16433: Hi, i made sone changes in curaengine then after building it i replace the curaengine .exe file in cura. Now i am trying to slice but this error pops out.

Neeraj729, 2023-08-09:
### Cura Version

5.4

### Operating System

windows11

### Printer

custum

### Name abnormal settings

i have replace curaengine.exe file with newly build file

### Describe your model

yes i have sliced the same model before

### Add your .zip here ⬇️

Hi, 
i made some changes in curaengine then after building it i replaced the curaengine.exe file in cura. Now i am trying to slice but this error pops out stating an unexpected error.
Can anyone help me on how to run modified curaengine.exe using cura?

MariMakes, 2023-08-09:
Hey @Neeraj729,

Welcome to the UltiMaker Cura Github 🚀 

I'm super curious. What do you want to achieve by changing your curaengine? 
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

As far as I understand, making changes in the curaengine, is a big ordeal and requires you to run from source. We've documented how to do that here: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

Good luck!

Neeraj729, 2023-08-09:
Hi,
I am doing research on concrete 3D printing, for which I need some
modifications in G code, I know the logic behind the change to be done in
curaengine. But I am not able to run cura engine.
So, what I can understand is that I have to run cura also from source code
to run modified curaengine code. Is there any to run it from cura GUI?

Regards
Neeraj Singh

Neeraj729, 2023-08-09:
> Hey @Neeraj729 <https://github.com/Neeraj729>,
>
> That sounds pretty cool.
> What you are trying to do is not a workflow we actively support. 😅
>
> I double-checked with the developers and they confirmed that if you build
> the CuraEngine.exe correctly, you should be able to overwrite the
> C:\Program Files\UltiMaker Cura 5.4.0\CuraEngine.exe. So there might be an
> error in your version of the engine.
>
> I don't have experience changing Engine code, but I believe what you are
> describing is something that ThomasRahm has done to distribute his code
> here. perhaps his steps make sense to you
> https://github.com/ThomasRahm/CuraEngine

Hi,
Thank you so much for looking into this for me, I really appreciate it.
Once I finish my research, I will definitely share my results and methods
with you.
Till then if required I will bother you.😅

Regards
Neeraj Singh

ThomasRahm, 2023-08-09:
You can use https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source#5-connecting-to-cura to start the UI to enable it to connect to an external engine. 
Instead of the `cura_app.py`  you can just use the `Ultimaker-Cura.exe` of your regular Cura Installation.
The port may be different, you will need to check the Cura log files (sometimes it prefers using another port, i suspect windows doing something weird)
This makes it much easier to develop, as you can just set parameters in your IDE, and then you can start CuraEngine from your IDE. If you got your IDE to compile CuraEngine, the hardest part of setting your dev-setup up is already done.

P.S. The Engine should work if you just drop it in, but i remember i had other dll files that the compiler copied into my build directory that i had to drop in there too. I dont know if this is till the case in 5.4 though.

Neeraj729, 2023-08-14:
Hi,
I tried to slice a simple geometry using cura from source but the gcode
generated doesn't seem right, i am attaching a gcode file. What could be
the reason behind this output?
Can you help me in understanding what is going wrong and why am I not able
to create gcode using the command line.

Thanks &
Neeraj Singh

Neeraj729, 2023-08-16:
Hi,
I am creating a small concrete printer for research work in which I want to
rotate the nozzle head in the direction of movement.
I guess I have to make changes in the slicer code to write gcode to control
this additional motor. I am trying to build cura from source but having
trouble executing it. Can someone guide me for this

Regards
Neeraj Singh

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant