Ultimaker/Cura #16434: External brim on supports

mikeywaldron, 2023-08-09:
### Is your feature request related to a problem?

Tree supports pull away from the build plate is a nightmare, especially when so much has went into organic supports 

### Describe the solution you'd like

I've found using the support brim along side the model brim and unsticking brim on outside only, allows the brim to fully secure down the tree support as it does internal and external of the tree

### Describe alternatives you've considered

I've tested this on multiple prints and running both brims works out good. I would like to see this in the support section, as apposed to having to run a full brim on the models in question, allowing full brims of the trees and using a skirt for the main file

### Affected users and/or printers

All i image.

ThomasRahm, 2023-08-09:
Increasing `Initial layer diameter` also increases the area (if support brims are used) that the branch has on the build-plate to improve adhesion. A branch can never be larger than `Trunk Diameter`, so if you increase your `Initial layer diameter` ensure that the `Trunk diameter` is smaller than it.

mikeywaldron, 2023-08-10:
I mean partially there. it appears just to make the initial size wider. The Inside and outside brim sandwiches the trunk so there is zero movement. using the Support brims have failed for me a lot using brim and having the brims external to the trunk seems a lot more secure. would be nice to see this as a option