Ultimaker/Cura #16435: nozzle bug for kp3s pro

PrimusMaster, 2023-08-09:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Kingroon kp3s pro

### Reproduction steps

1. Add a kp3s pro printer from kingroon
2. Trying to select a nozzle for the printer configuration

### Actual results

When you select the nozzle selection in the pla material it says empty

### Expected results

I would expect being able to select a nozzle for my printer, i had to change my printer for the kp3s normal instead of the pro

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2023-08-10:
Thanks for the report.
The printer definition files come from the manufacturer or from a member of the community.  UltiMaker can't be responsible for other companies products or how they are supported.  I'll remove the bug label and close this.  You will have to check Kingroon to see if they have updated files for your printer.  Maybe there is a group on Reddit you could aski.

closed 2023-08-10 by GregValiant

thebitsworld, 2023-09-20:
I've the same problem with my Kingroon, change printer type to Kingroon KP3S to make it work, this is a dirty fix, waiting for update version of cura