Ultimaker/Cura #16439: Enable Support Interface grayed out

AndreKR, 2023-08-09:
### Cura Version

5.3.0

### Operating System

Windows 10

### Printer

Flsun QQ

### Reproduction steps

I'm not exactly sure how I ended up in this state. *Maybe* just by disabling "Enable Support Interface" while its children were enabled?

### Actual results

"Enable support interface" is grayed out.

The tooltip says "This setting is not used because all the settings it influences are overridden." (But I'm not sure if this text is actually relevant to the problem.)

### Expected results

I expect it to be not grayed out so I can enable the support interface.

### Add your .zip and screenshots here ⬇️

[image]

AndreKR, 2023-08-09:
Here's `%APPDATA%\Roaming\cura\5.3\user\flsun_qq_extruder+%232_user.inst.cfg` in case that's relevant:

```
[general]
version = 4
name = flsun_qq_extruder #2_user
definition = flsun_qq

[metadata]
type = user
setting_version = 21
extruder = flsun_qq_extruder #2

[values]
brim_gap = 0.3
infill_pattern = grid
infill_sparse_density = 30
speed_support_interface = 50
speed_topbottom = 40
speed_wall_0 = 40
speed_wall_x = 50
support_angle = 70
support_bottom_enable = False
support_infill_rate = 5
support_offset = 0
support_roof_enable = False
support_z_distance = 0.2
wall_overhang_angle = 45
wall_overhang_speed_factor = 50
```

GregValiant, 2023-08-10:
That happens when both the roof and floor interfaces are turned off.
Hover the mouse over the "Support" heading bar and an icon will show up on the right end.  It sort of looks like 3 slide switches.  Click on that and in the support settings makes the Support Interface settings visible so you can adjust them.
Let us know if that's what it was.

AndreKR, 2023-08-14:
Nope, that's not it, I have them all visible.

I found a way to deliberately create the problematic state by using the curled reset arrow:

https://github.com/Ultimaker/Cura/assets/1188538/02a49c8f-e9d3-4736-a506-a245994e3bd1

I don't remember why I would click that reset arrow, so I'm not sure if this was the exact way I initially got into this state. (I think I rather switched the profile, but not sure.)

AlansCodeLog, 2023-09-06:
Am also getting this, my workaround was to go to another profile that didn't have them on, but let me turn them on, then switch back to the profile I wanted and keep changes when prompted.

GregValiant, 2023-09-06:
You can see that "Enable Support Roof" and "Enable Support Floor" are indented below "Enable Support Interface".  They are "children" of the "parent" setting "Enable Support Interface".  If you happen to set the Roof or Floor differently than the main setting they become un-linked and will stay that way unless you re-link them with the round arrow icon.
A lot of those settings are cross-linked.  I'm not saying that there isn't a problem in there someplace but usually with some fooling around you can get them adjusted as you want.

The QT controls that Cura uses are "ambidextrous" and can be made to work with Linux, MAC, and Windows.  Those controls are not really designed for a specific system and that's why Cura uses them.  You only have to aintain a single interface for all three operating systems (and Chrome and Android)
We used to joke in the design shop I worked at that "Ambidextrous means it won't work either way.".  It was something all the customers wanted and we were loathe to give them.  There would always be problems down the road if you tried to make a single design work for everything and no one ever wanted to pay for the re-work.

AlansCodeLog, 2023-09-06:
I understand they are children, the problem is you cannot "relink" them again, they disappear and *cannot* be made to appear again without the weird workaround I mentioned.

I don't think this is a QT problem, but a logic problem.

Even if I have overridden and unlinked both children properties, my first expectation is I should still be able to override the parent. I don't think the "This setting is not used..." error should happen on a parent if it's children are overriding it (also the error could be a bit more verbose about what is overriding it). 

Or if that is the intended behavior, it's children should not disappear if the parent is in such a state. 

Also I do not understand why the children are calculated properties in this case. What the are calculated *from*? The parent?

GregValiant, 2023-09-07:
This is from fdmprinter.def.json:
support_interface_enable
"enabled": "support_enable or support_meshes_present"
support_roof_enable
"enabled": "(support_enable or support_meshes_present) and support_interface_enable"
support_bottom_enable
"enabled": "(support_enable or support_meshes_present) and support_interface_enable"

So it would appear that disabling  "Generate Support" might be having an effect.  It's also possible that something in the code isn't getting the memo when the setting is changed as maybe it isn't broadcasting the change to other functions until something else is tripped and the message goes out.  At that point everything would be playing catch-up and you can get some odd behavior.

EliW, 2025-02-01:
Just wanted to come in and confirm that I just ran into this bug myself. And though exactly the steps it took to make it happen I was playing around with a lot of different settings moving to Cura from another program.  I had played with using the support interface options and deciding they weren't what I needed. I went to turn them off. I ended up turning off the floor and then turning off the ceiling independently afterwards I immediately thought "that was silly. I can just click the disable support interface toggle after doing so that toggle grade out just like I'm here and is now impossible for me to turn back on.

GregValiant, 2025-02-02:
This report is for Cura 5.3.0.  @EliW what version of Cura did you notice the problem in?

EliW, 2025-02-03:
> This report is for Cura 5.3.0. [@EliW](https://github.com/EliW) what version of Cura did you notice the problem in?

5.9.0 on MacOS

EliW, 2025-02-04:
> unless you make the child settings visible

Is there even a way to do this @fieldOfView ?  Without loading an old profile?  I've been trying to look for a way and haven't found it.

EliW, 2025-02-06:
@fieldOfView ... I don't see how to do that:

So here is the current section of the print settings showing the gray button I can't interact with:
[image]

If I go into the setting visibility pane for preferences, everything is marked as being visible and toggling things on or off doesn't change anything:
[image]

Hovering over the information, mark says that it's been hidden because of the setting above:
[image]

NOTE: yes, generate support is turned on
[image]

And if I try to use the search bar in the settings to find the floor or roof settings, it doesn't show up either ... However, the parent in Naple support interface Gray option does appear in the filter:
[image]

So again, there doesn't seem to be a way I can see at all to make those options appear again, and to no longer have enable support interface be gryed out.

Vanhel86, 2025-03-17:
Found some decision for fix that problem, but it's not simple. I use this instructions to modify *.curaprofile https://community.ultimaker.com/topic/21501-is-it-possible-to-editcreate-curaprofiles-in-a-text-editor/?do=findComment&comment=200179
I deleted all rows in file profile with "support_interface_*".
After that Enable Support Interface, available again