Ultimaker/Cura #16441: Benchy failing to Slice

Aughard, 2023-08-09:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Artillery Sidewinder X2

### Name abnormal settings

Default

### Describe model location

Central facing East

### Describe your model

Basic Benchy which I have printed many times

### Add your .zip here ⬇️

[ASX1_3dbenchy.zip](https://github.com/Ultimaker/Cura/files/12305438/ASX1_3dbenchy.zip)

[cura.log](https://github.com/Ultimaker/Cura/files/12305447/cura.log)

Aughard, 2023-08-09:
I ge tthe error message 
Slicing failed
Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

GregValiant, 2023-08-10:
Thanks for the report and file.
Where did you get that model??  It's full of errors and I've never seen a Benchy that bad.  I exported the model from Cura to try to fix it and it's 58mb.

All the red surfaces you see here are inverted so that the inside is facing outside.  The repaired model is on the right.
[image]

Since you've tried the normal fixes for this bug, try repairing the model and let us know if it slices.

I'll leave the bug label on this for now but this could definitely be a model problem.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic