Ultimaker/Cura #16442: Tree support fails slicing on round object

db-puzzle, 2023-08-09:
### Cura Version

5.4.0

### Operating System

MacOS Ventura

### Printer

Ender3 V2

### Name abnormal settings

Tree support crashes, Normal support structure works.

### Describe model location

Can't move it due to shape.

### Describe your model

Round vase with circular overhang.

### Add your .zip here ⬇️

[](url)
[Slicing Failed.3mf.zip](https://github.com/Ultimaker/Cura/files/12305875/Slicing.Failed.3mf.zip)

GregValiant, 2023-08-10:
Thanks for the report and the file.
I didn't have any trouble slicing this one.  There is a well known bug that can keep models from slicing.  Try moving the model on the build plate, rotating the model, or change the Initial Layer Height by a little (like from 0.20 to 0.23).
[image]
I'll mark this as a duplicate.  The Cura team has a lot of these slicing errors to look at so I'll go ahead and close this as well.  I hope you understand.

closed 2023-08-10 by GregValiant

db-puzzle, 2023-08-10:
Indeed moving the model on the baseplate solved the problem.

Thank you very much for the reply and congratulation on the fantastic improvement on tree supports!

Danilo