Ultimaker/Cura #16443: help

ngroborocks, 2023-08-09:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Cobra Plus

### Name abnormal settings

no

### Describe model location

middle

### Describe your model

not sucsessfull slice

### Add your .zip here ⬇️

[New folder.zip](https://github.com/Ultimaker/Cura/files/12306227/New.folder.zip)

ngroborocks, 2023-08-09:
help

MariMakes, 2023-08-10:
Hey @ngroborocks,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Thanks for the Project File 👍
I took a look at it in Cura 5.4 and I see that your Maximum Branch Angle and Preferred Branch Angle have very extreme values. 
If I restore them to the default settings the slice is succesful:
[image]

Can you help me understand why you believe these settings should be this low?
To me, it doesn't seem like a bug in Ultimaker Cura anymore, since your setting values are extremely low. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-08-24 by github-actions[bot]