Ultimaker/Cura #16445: fail in the slicing

3dimpresionesesp, 2023-08-09:
### Cura Version

5.4

### Operating System

windous 11

### Printer

anycubic kobra

### Reproduction steps

I put the stl in to the program
I press slicing
and the fail appear in my screen

### Actual results

the fail appear

### Expected results

I expect to have the stl slicing in to the g.code to can print the object I dising

### Add your .zip and screenshots here ⬇️

[AK_RSP GARAJE (8).zip](https://github.com/Ultimaker/Cura/files/12307214/AK_RSP.GARAJE.8.zip)

3dimpresionesesp, 2023-08-09:
pls solve it as fast as posible thanks

GregValiant, 2023-08-09:
Thanks for the report.
There is a bug that can keep models from slicing.  Have you tried moving the model on the build plate or rotating the model?  Sometimes changing the Initial Layer Height by just a little can fool this bug.
But your model has errors as well and needs to be repaired or it may not print correctly.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 496 Non-manifold edges (?)
--> 648 Inverted faces (?)**
--> 0 Degenerate faces (?)
**--> 63 Duplicate faces (?)**
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 10164ms ------
-> Vertex count changed from 8994 to 9456 (+462)
-> Triangle count changed from 19076 to 20138 (+1062)
You are running windows and MS 3D Builder has a pretty decent repair utility.

I'll mark this as a duplicate and go ahead and close this.  The Cura team has lots of examples of this behavior and I didn't have any trouble slicing the model.  It's possible your settings are causing a problem but without a project file I can't tell.
[image]

closed 2023-08-09 by GregValiant