Ultimaker/Cura #16446: Slice Failure

thocan, 2023-08-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Sovol SV06

### Name abnormal settings

N/A

### Describe model location

Rotating and moving file does not help

### Describe your model

Haven't sliced successfully before. Did mesh fix, same result. File came from popular creator with no issues reported.

### Add your .zip here ⬇️

[Bug Report File.zip](https://github.com/Ultimaker/Cura/files/12308153/Bug.Report.File.zip)

mikeywaldron, 2023-08-10:
Try Unticking Print Thin Walls and see if it slices then

thocan, 2023-08-10:
That did it, going to fire off a print and see how it comes out, but it did successfully slice. Thanks!

closed 2023-08-10 by thocan