Ultimaker/Cura #16448: program errors during slicing

hymnatr, 2023-08-10:
### Cura Version

5.4

### Operating System

windows 11

### Printer

craftbot flow idex

### Reproduction steps

added stl file to project.  tried to slice and it fails.  Works on version 5.0
I am unable to get any files  to slice on 5.4

### Actual results

nothing

### Expected results

should get the option to save the sliced file

### Add your .zip and screenshots here ⬇️

[CFI_knurled 5-8x24 thread insert v1.zip](https://github.com/Ultimaker/Cura/files/12311163/CFI_knurled.5-8x24.thread.insert.v1.zip)

GregValiant, 2023-08-10:
Thanks for the report and the file.
There is a bug in 5.4 that can keep models from slicing.  There are three workarounds that often fool the bug.
Move the model on the Cura build plate (that worked for me on this one).
Rotate the model in Cura (even though it's round it can make a difference)
Change the Initial Layer Height by just a bit (like from .20 to .23).

I'll mark this as a duplicate and the Cura team will take a look.  Cura 5.4.0 on Win11 seems to have issues.  You might want to use 5.2.2.  You don't have to un-install 5.4.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic