Ultimaker/Cura #16449: model within size wont slice because "to large"

TheNecronomica, 2023-08-10:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Pro

### Reproduction steps

1. This model: https://www.printables.com/model/180177-vase-mode-trash-can
2. Load model into Cura, stand on end and center it.
3. Change size to 150%
4. model is now 219x219x267mm
5. printer settings in Cura has build plate at 220x220x270mm
6. slice failed because model is out of bounds.

### Actual results

slice failed due to size even though model is smaller than build plate.

### Expected results

slice to function and load normaly

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/12311395/cura.log.zip)
[Vase Mode Trash Can Bug.zip](https://github.com/Ultimaker/Cura/files/12311453/Vase.Mode.Trash.Can.Bug.zip)
[image]
[image]
[image]
[image]

GregValiant, 2023-08-10:
Thanks for the report.
When you enable "Avoid Printed Parts When Traveling" the travel moves avoid the part by traveling outboard.  That is expected behavior.

I'll remove the bug label and close this.  I hope you understand.
[image]

closed 2023-08-10 by GregValiant

TheNecronomica, 2023-08-10:
There are however no travel moves in this print

GregValiant, 2023-08-10:
But the Cura Engine doesn't know that until after the slice.  It must allow for the possibility that the extra margin will be required.  I understand that Spiralize is enabled and that means _Avoid Printed Parts when Traveling_ is not required.

TheNecronomica, 2023-08-10:
so then why would the non existent travels stop it from even attempting to slice?
i would understand if it tried to slice and noticed travel moves that went outside the plate and then gave a warning.
but to not even attempt to slice because of the possibility of traveling outside the plate, does not make sense.