Ultimaker/Cura #16450: Baby Groot Alex Dot holder

Jimjam-puddleduck, 2023-08-10:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

No

### Describe model location

Have tried repositioning, same issue occurs

### Describe your model

1st time using. It Baby Groot Echo Dot Head holder. Xray view seems sound

### Add your .zip here ⬇️

[Baby_Groot_for_Echo_Dot___Amazon_Alexa_2966407.zip](https://github.com/Ultimaker/Cura/files/12311713/Baby_Groot_for_Echo_Dot___Amazon_Alexa_2966407.zip)

mikeywaldron, 2023-08-10:
See if print thin walls is enabled, if it is try to disable and re slice?

GregValiant, 2023-08-10:
Thanks for the report.
Both models have minor errors but even after repair I can't get them to slice.

The Cura team will take a look.

Jimjam-puddleduck, 2023-08-10:
> Now I'm totally lost.
> After fooling around with a couple of other bug reports I went back to
> this and the repaired file sliced with my Ender 3 Pro as the active printer
> with my normal settings. Print thin walls was not turned on but it wasn't
> the first time when it wouldn't slice.

Yeah I just tried the print thin walls to no email.
I'm still new to printing to be honest so wasn't sure if it was something I
was doing wrong.

Jimjam-puddleduck, 2023-08-18:
Hi,

Was there any further update on this issue?

Many thanks.

GregValiant, 2023-08-19:
If you go [HERE](https://github.com/Ultimaker/Cura/releases?page=2) you can scroll down and download Cura version 4.13.1.  You can install it and there is no need to un-install other Cura versions.  It isn't near as fussy about models as 5.4.0 is.  Of the 5.X Cura versions 5.2.2 is probably the best of the lot.