Ultimaker/Cura #16451: Strange Z seam generated during vase mode (spiralize outer contour) slicing

baronvonk, 2023-08-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Artillery sidewinder X1

### Reproduction steps

When slicing a **conical shape with vase mode**, a strange z seman appears. It would be no problem if it wouldn't be visible on the actual print.

Selecting "smooth spiralized contour" doesn't help.
Only happend on conical-ish shapes, not on cylinders. 

Doesn't happen with other slicers like ideamaker.

Layer height: 0,3
Initial layers: 0

### Actual results

Visible "Z seam" type of surface imperfection

### Expected results

Smooth spiralized path with no irregularities

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]

GregValiant, 2023-08-10:
Thanks for the report but without a project file ("File | Save Project" then zip) it's not possible to investigate this.

baronvonk, 2023-08-11:
> Thanks for the report but without a project file ("File | Save Project" then zip) it's not possible to investigate this.
Thanks for your reply. I added the project file, sorry for that.

GregValiant, 2023-08-11:
Thanks for the file.
I can see the problem in the Cura preview.  I saved the Gcode and read it into AutoCad and those little spots aren't in the gcode so I think this is a display problem.
If you look at the inside the extrusions don't have a matching jog in them and if it were a problem with the path then they should show an irregular movement on the inside as well.
Here is my Cura preview with the spots.
[image]

This is a view of the same area in AutoCad.  I know it's hard to make out but there definitely aren't any jig-jogs in the code.
[image]

I haven't seen any evidence of that in my prints either so I'm unable to duplicate the problem.  Have you been seeing the issue turn up in your prints?

If you will post a gcode file I can check, but right now it looks like a display issue rather than a slicing issue.

baronvonk, 2023-08-12:
Thank you for investigating. I'm sure it's not just a preview problem, because I can actually see the toolhead making a weird "knock" movement where this strange Zseam shows on cura. Like you can see on the picture I posted on the first post, this is a physical artefact that is visible on prints.

While I was creating several STLs with different shapes and their associated gcodes, I noticed the "seam" is more present when the slope of the cone is greater. It totally disappears when printing a cylindrical section. Si I created a single STL with different slope angles. 

[image]

I loaded the Gcode in another slicer (ideamaker) and saw a visible discontinuity where the artefacts appears.

[image]
[zseam vase mode.zip](https://github.com/Ultimaker/Cura/files/12327140/zseam.vase.mode.zip)

Here is a STL that has different slopes angles, and the associated Gcode sliced on Cura 5.4.0, along with some screenshots and pictures of the test print.

I might add that I use Klipper, printing at 100mm/s with pressure advance. This little imperfection might be amplified by pressure advance. But in any case, this artefact isn't supposed to be there in the first place I suppose.

Let me know if you need more info, I'd be happy to help as much as I can. This artefacts heavily impacts the project I'm working on.

baronvonk, 2023-08-12:
Added info:
Here is the result of this [online gcode viewer](https://gcode.ws/)

[image]

There is a strange offset going on here at each "layer".

This offset is greater with lower slope angle 
[image]

baronvonk, 2023-08-12:
This offset seem to be applied at the start of each layer. I'm not too familiar with gcode so I might be wrong here, but this 1mm gap seems strange on a spiralized contour.

```
G1 X158.152 Y205.433 Z1.346 E0.12902
G1 X157.097 Y205.48 Z1.348 E0.1304
G1 X156.048 Y205.507 Z1.349 E0.12957
**G1 X155** Y205.517 Z1.35 E0.12941
;TIME_ELAPSED:46.419073
;LAYER:2
M106 S234.6
;TYPE:WALL-OUTER
;MESH:cone to cylinder.stl
**G1 X155** Y205.248 E0.03322
**G1 X153.956** Y205.238 Z1.352 E0.12892
G1 X152.912 Y205.212 Z1.353 E0.12895
G1 X151.862 Y205.166 Z1.354 E0.12978
G1 X150.824 Y205.102 Z1.355 E0.12841
```

smartavionics, 2023-08-12:
Could you please provide a project (.3mf) file for that zseam test part so we can see the settings. Thanks.

baronvonk, 2023-08-12:
There is already a 3MF file with the settings at the end of the initial post "TEST siralized cone - strange Z seam.zip"
 It uses the same settings as the multi angle test print.

baronvonk, 2023-08-12:
> Could you please provide a project (.3mf) file for that zseam test part so we can see the settings. Thanks.

There is already a 3MF file attached at the end of the initial post. "zseam vase mode.zip". It is a conical shape, but uses the same settings as the multi-angle slope test print posted after that.

smartavionics, 2023-08-12:
Ah, yes, I've got that and tried it out. So, using my Cura (https://github.com/smartavionics/Cura/releases), I find that without enabling the smoothing spiralized contours setting, I see artifacts around the z-seam. With that option enabled, the artifacts go away. That's what that setting is designed to do.

Without the smoothing I see this...

[image]

With the smoothing...

[image]

baronvonk, 2023-08-12:
Thank you for the feedback @smartavionics but you are using an older version of cura. I tested my STL files on an even older version (4.13.1) and the problem is still here, no matter if spiralized contour is enabled or not.
Also, based on your screenshots, it seems like you are slicing a cylindcrical shape and not a conical one. The issue i'm getting is not about vertical cylindrical shapes, as I initially mentioned.
About the smooth spiralized contour setting: I already tested this setting since it is the most obvious one, without any result. This is specified in the initial post.

smartavionics, 2023-08-12:
Yeah, but my Cura variant diverged from UM Cura before 4.13.1 so it is possible that the behaviour is different. Anyway, using your test model, I still get good results when the smoothing is enabled.

[image]

I know this doesn't really help you if you want to use UM Cura.

baronvonk, 2023-08-12:
@smartavionics Thank you for your very quick response.
As I previously said, sadly this setting isn't hepling at all: 

[image]
[image]

I didn't realize that the cura variant you where talking about was your own variant. I'll test it and give you feedbakc as soon as I can. Thank you.

baronvonk, 2023-08-12:
@smartavionics I installed the Cura-4.20.14-mb-amd64.exe file from your repo and here are the results:
[image]
I guess something is wrong here.

smartavionics, 2023-08-12:
The Cura layer display has some inherent limitations that stop it from displaying spiralized layers correctly. The only way to visualize the result is to use some other g-code viewer. I use the craftware slicer to look at gcode produced by Cura.

GregValiant, 2023-08-12:
I just printed the lower portion of one of the towers and I didn't get that defect in the print.  I can't get a good image with my camera, but there isn't any thing to show.  The "z seam" side looks like the rest of the print.  I used 5.4.0 (Win10) for the slice.

Here is the preview of the Zseam side in IdeaMaker.  It's how the print looks as well.
[image]

(Older fork or not, I always use the @smartavionics version for spiralize.)

baronvonk, 2023-08-12:
Okay, your cura variant seem to handle it pretty well! In this variant the "smooth spiralize outer contour" actually makes a difference. Here are some screenshots:
with smooth spiralized outer contour (random bed & layer adhesion are new issues that are easy to overcome):
[image]

without smooth spiralized outer contour
[image]

So it basically works. And it shows how the "smooth spiralize outer contour" is'nt relevant anymore on UM cura for this specific problem.

Your last feedback lost me.

> I used 5.4.0 (Win10) for the slice
> I always use the @smartavionics version for spiralize.

What cura version did you actually use to get this result?

baronvonk, 2023-08-12:
The thing is, I'm not really interesed into the _"I sliced it and it went fine"_ thing because it really doesn't help me at all. I'd like to understand what is wrong with my **perfectly official cura release** and why all of my vase mode prints look like absolute crap unless I print a perfectly vertical cylinder. Any alternate random version of Cura cannot be considered as a valid point.

baronvonk, 2023-08-12:
solved...

closed 2023-08-12 by baronvonk