Ultimaker/Cura #16454: all Versions upper then 4.13 do not work on my computer

GunnMan, 2023-08-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

anycube

### Reproduction steps

All versions since 4.13 are not usable !!!
- Key and mouse commands take ages to react
- I am getting mad , so I still work with Version 4.13
This behaviour is not known from any other Programm I have installed

### Actual results

No Help, no remark, so I can not do anything against it

every action either by keyboard or mouse needs at least 6 seconds for seeing a result

### Expected results

after klicking, I expect an reaction pretty soon

time and delay's will not get documented by pictures

### Add your .zip and screenshots here ⬇️

[image]
[image]

MariMakes, 2023-08-11:
Hey @GunnMan,

Welcome to the UltiMaker Cura Github 🚀 
I'm sorry to hear you are having so much trouble with your Cura 😞 

To learn better about what's slowing your software down, could you install a plug-in that performs more extensive logging? 

You can download it here: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
After it's installed can you.... 
	1. Select Extensions
	2. Select Remote Support
	3. Start CPU profiler
	4. Repeat your steps to reproduce. 
	5. Stop CPU profiler
	6. Select Save Remote Support Archive
	7. Go to the Marketplace
	8. Select the gear icon 
	9. Uninstall Startup Optimiser
	10. Share the Remote Support Archive Zip file here on Github

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

GunnMan, 2023-08-26:
I already have sent an email:

Betreff: | Re: [Ultimaker/Cura] all Versions upper then 4.13 do not work on my computer (Issue #16454)
-- | --
Mon, 14 Aug 2023 22:15:43 +0200
Gunnar Quincke <gunnar.quincke@quincke-luten.de>
Mariska <notifications@github.com>

Hello Mariska,

I think I have Installed [ExtensiveSupportLogging](https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging),

But in Cura there is no (2. Select Remote Support)

I can not start Remote Support

even there are no updates available (Update-Prüfer

MariMakes, 2023-08-28:
Hey @GunnMan,

To double-check if you have the plugin available you can execute the following steps:
1. Click on the marketplace
2. Click on the Gear
3. Read through the list of installed plugins

[image]

Then it should show up like this
[image]

Can you doublecheck if the software is installed correctly?

GunnMan, 2023-08-28:
[remote-support_23-08-28-21_22_37.zip](https://github.com/Ultimaker/Cura/files/12457679/remote-support_23-08-28-21_22_37.zip)

I also answered via email, 
Thanks for your try to help.

The Startup Optimiser was and is not installed.

every action to load  a sample and did your advised task took at least 10 seconds for every move :(

let me know if you can fix this 

regads Gunnar Quincke [GunnMan]

:)

[remote-support_23-08-28-21_22_37.zip](https://github.com/Ultimaker/Cura/files/12457717/remote-support_23-08-28-21_22_37.zip)

GunnMan, 2023-08-28:
I am a bit stupid with this exchange , I still need your further advice to solve my problem, did I closed the Item ??

MariMakes, 2023-09-04:
Hey @GunnMan,

This issue is very much outside my field of expertise 😦😖
I'll bring it up to the team to see what they can see in your logs.

Fingers crossed 🤞