Ultimaker/Cura #16458: Learn More link points to salesforce

oohookoo, 2023-08-11:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

ender 3 v2 neo

### Reproduction steps

Marketplace-> manage packages link -> top of window has a notification about making sure to keep plugins up to date.
'Learn More' directs to a salesforce login page 

#unexpectedbehavior.

### Actual results

directs to salesforce page

### Expected results

doesn't go to salesforce login.

### Add your .zip and screenshots here ⬇️

[image]
[image]

MariMakes, 2023-08-11:
Hey @oohookoo,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

We are migrating our support pages, so some of our redirects are broken. I encountered exactly the same issue earlier this week and reported it. I was told it was resolved but seeing your report, it might not be the case. 

I'll bring it up to the team to see what they can do to fix it. Thanks for keeping us sharp 😊 
Fingers crossed 🤞

MariMakes, 2023-08-11:
Quick update, there was a typo we overlooked https://github.com/Ultimaker/Cura/blob/298f00ebdeb545ea93d56001c962b5eb8a26f1ab/plugins/Marketplace/resources/qml/ManagedPackages.qml#L17

So we are double-checking the work done. 
Hopefully it's resolved soon. Thanks again 👍

MariMakes, 2023-08-11:
Hey 👋 
Because of your report a number of Learn More Links have been fixed now, 👍 
I'll close this issue, thanks again 😊

closed 2023-08-11 by MariMakes