Ultimaker/Cura #16459: Zero 300mm body failure

ProfMoon, 2023-08-11:
### Cura Version

5.3.1

### Operating System

MacOS Ventura

### Printer

FLSun V400

### Name abnormal settings

None.  I used the same settings I've used successfully in previous prints.

### Describe model location

I have tried rotating the model, but it continues to fail.

### Describe your model

It is a 300mm version of Zero's body, from Nightmare Before Christmas

### Add your .zip here ⬇️

[Zero_Body_300mm.3mf.zip](https://github.com/Ultimaker/Cura/files/12318475/Zero_Body_300mm.3mf.zip)

ProfMoon, 2023-08-11:
I continued tweaking the rotation and position, and finally got it to complete slicing.  Not sure what happened with the previous attempts...

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic