Ultimaker/Cura #16461: Make "Spirilize Outer Contour" Compatible with modifier messhes

B-Man1, 2023-08-11:
### Is your feature request related to a problem?

I often want to print the bulk of my prints in vase mode, which caps off the bottom surface, but does not cap off the top.  I tried to solve this in cura using modifiers but spiralize outer contour did not show up in the list of things I could modify, and I was not able to modify a part that had spiralize outer contour turned on.  Seems like I can either spiralize all of it, or none of it.  It would be great to be able to spiralize some of it

### Describe the solution you'd like

Make "Spiralize Outer Contour" Compatible with modifier meshes, Or, adding the ability for spirilize outer contour to cap off the top of parts

### Describe alternatives you've considered

 I tried to solve this in cura using modifiers but spiralize outer contour did not show up in the list of things I could modify, and I was not able to modify a part that had spiralize outer contour turned on. 

### Affected users and/or printers

All users all printers

GregValiant, 2023-08-11:
I've done this several times.  I'll explain a little bit.
You have a vase and you want a conventional base and spiralized upper portion.  You add a "Pause at Layer" at a convenient place for the transition.  As long as you keep the layer height the same it will work (actually, you can change the layer height but then the Pause will be at different layers...not a real problem but more difficult).  The line width can be different if you like.

Slice it twice and create the two gcodes.  When you merge the gcodes the PauseAtHeight section becomes your transition.  Keep everything above the pause command in the base file, and throw away everything starting with the pause command down through the end of the file.
In the second file, copy everything starting at the line below the pause command and paste it into the base file.
You need to check that the X, Y, Z, and E values are going to merge nicely, but they will be close.  The final check is that if there was a retraction at the end of the base file but there wasn't one at the continuation of the spiral file, you need to account for that as it may need a prime to get the filament back to the tip of the nozzle.
It's more nervous than difficult but with a bit of practice the results can be excellent.

By it's nature, spiralize either "is" or it "isn't" so modifier meshes will likely never be able to turn it on and off.

B-Man1, 2023-08-11:
Seems like it would be fairly straight forward to incorporate this into a modifier mesh, or simply add a "Base thickness" and "Cap thickness" option so that vase mode prints could be capped at both ends by solid infill instead of just at the bottom

tripletto, 2023-08-19:
Love to see that feature implemented as well!

mateoescalanter, 2025-04-24:
Hi, a bit late, but this tool I made would work for what you were looking for [https://github.com/mateoescalanter/gcode_mixer](https://github.com/mateoescalanter/gcode_mixer) I am leaving it here for anyone who may need it in the future.