Ultimaker/Cura #16464: Slicing failed for a lithophane

Klemmie, 2023-08-12:
### Cura Version

5.4.0

### Operating System

MacOS Ventura

### Printer

Creality Ender-3 Pro

### Name abnormal settings

Custom settings: Layer height 0.1mm, initial layer height 0.2mm, 100% infill, 80mm/s print speed, 30mm/s initial print speed, retraction on at layer change and at 50mm/s speed, 6.0mm minimum extrusion distance window, 

### Describe model location

Center of the build plate, "standard" orientation

### Describe your model

It's part 4 of a 9 part lithophane all generated using the same source. The other 8 parts slice perfectly fine with these settings which is why I don't want to change everything for this one part. This specific part is the thickest of all 9, so I'm guessing this is why the slice is failing

### Add your .zip here ⬇️

After some fiddling around I found that changing the initial layer height to 0.21 or basically anything other than 0.# (so adding more digits after the first number) makes the slice work, quite annoying as this influences the quality of this specific part relative to the other 8, but this is something I can live with. Not exactly sure why this would be an issue though.

[Lithophane_slice_failure.3mf.zip](https://github.com/Ultimaker/Cura/files/12327108/Lithophane_slice_failure.3mf.zip)

Klemmie, 2023-08-12:
My work around is to slice at 0.19 initial layer height and then edit the gcode manually to set it to 0.2 and update the next layer heights to accommodate this change. I haven't printed this file yet, but I can't see why this won't work

bacon4bfast, 2023-08-30:
Happened to me too. When I set the infill to 100% the number of top layers goes to 0 and the number of bottom layers goes to 999999. I turned down the bottom layers to 4 and was able to slice fine.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic