Ultimaker/Cura #16466: "Show Configuartion Folder" does nothing

onkytonk, 2023-08-12:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Fresh installation of windows. I hadn't even restored my backup yet. 

Clicked Help> Show configuration folder

And nothing happens. 

### Actual results

Nothing. I manually have to go to the folder.

### Expected results

Config folder opens.

### Add your .zip and screenshots here ⬇️

N/a

GregValiant, 2023-08-12:
Thanks for the report.
It doesn't work for me either (Windows 10) so I can confirm this in 5.4.0.
I was looking through past issues and it seems that the "Show Configuration Folder" breaks once in while.  It appears to be broken again.

@MariMakes has this come up before in regards to 5.4.0?

MariMakes, 2023-08-15:
Hey 👋 

Thanks for tagging me 🙏 this was not on my radar. 😮 
However, I have no trouble opening my configuration folder 😅 

[image]

I'll bring it up to the team to see if they know what might be up💪

GregValiant, 2023-09-03:
I'm using Windows 10 Home Edition.  I hadn't noticed this, but "Show Configuration Folder" doesn't work on ANY of my installs of Cura.

MariMakes, 2023-09-04:
Hi @GregValiant,

I brought your issue up with the team. We asked several people to reproduce it but it always properly shows 🙈 
Not even when we emulate a fresh install by renaming the appdata/cura folder.

So we are a bit stuck troubleshooting this issue 😞 
We would like to keep it open in case someone has more information that we can work with.

GregValiant, 2023-09-04:
Thanks Mariska.
The fact that it doesn't work with the Creality 4.8 fork or with the SmartAvionics 4.20 fork sure points to it being a problem with my Windows Home installation.  It could be the way that "Qt.openUrlExternally(path);" is calling it.  There is no log entry at all and at this point we are officially beyond what I know about Cura.

MariMakes, 2023-09-11:
Hi @GregValiant,

I showed your analysis to the Developers and there were some lightbulbs in the room when they saw `Qt.openUrlExternally(path)`

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11026

Thanks for the report and the extra digging! 👍

GregValiant, 2023-09-15:
I've been going over old reports.  #6667 appears to be the same subject as this report.

DowntownTrainer, 2024-01-03:
Fresh install of Ubuntu and Cura on a newly configured dual boot machine (Win10 and Ubuntu 23.10) , having same issue.