Ultimaker/Cura #16467: Settings panel mouse wheel scrolling dampening / inertia seriously affects usability

tripletto, 2023-08-12:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Neptune 4

### Reproduction steps

Open Cura's print / material / speed etc. settings panel.
Try scrolling using the mouse wheel.
Observe the dampened / smoothed / kinetic scrolling effect.

### Actual results

Scrolling has a smoothing/dampening effect which hinders fast and accurate navigation.

**Impact:**
This behavior can impede users' ability to efficiently adjust settings, especially given the vast number of options available. This can lead to increased time spent navigating and potential user frustration.

**Suggestion:**
Consider reverting the scrolling behavior to match standard system-level mechanics or offer users the ability to toggle this behavior on/off based on their preference or over the ability to tweak the parameters of the dampening effect.

### Expected results

Scrolling should be responsive and consistent with typical system-level scrolling behavior, allowing for quick and precise navigation.

### Add your .zip and screenshots here ⬇️

Hello there ;)

GL17C4, 2023-08-15:
I'm having an issue with the scrolling inside Print Settings and I believe it is an issue with the mouse acceleration not being reset.
If I do multiple clicks of my scroll wheel in a row quickly, the settings page scrolls very fast. If I then do a single click on my mouse wheel, the page scrolls at the same speed that it had from the fast scrolling. If I do a second single click of my mouse wheel afterwards, it only scrolls a single line.
I am using different Logitech mice on two different computers with the Logitech G Hub software and both mice have the same behavior.

MariMakes, 2023-08-24:
Hi @tripletto,

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12110
We have a ticket on our backlog with the intent to improve this behavior.
From what I understand there have been a couple of tries to resolve this, but this is a waterbed problem. Once it's fixed for one type of mouse/trackpad on one Operating System it decreases the usability for another one. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12110

closed 2023-08-24 by MariMakes