Ultimaker/Cura #16469: Cura slicer engine crashes when I try to slice the model

f8lerror, 2023-08-12:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality3D Ender 3

### Name abnormal settings

no abnormal settings, no it does not slice, the sliceer engine crashes

### Describe model location

No it does not slice regardless of the models location

### Describe your model

This model has not sliced yet. It is not watertight. I have not tried that meshfix. The model in question is a set of cutters for clay that I made

### Add your .zip here ⬇️

[CE3E3V2_Surprising Vihelmo-Jarv.zip](https://github.com/Ultimaker/Cura/files/12327923/CE3E3V2_Surprising.Vihelmo-Jarv.zip)

Kremptor, 2023-08-13:
Hi,
There is a problem with your part, the middle one is not properly on the bed, try this corrected part

[CE3E3V2_Surprising Vihelmo-Jarv.zip](https://github.com/Ultimaker/Cura/files/12330659/CE3E3V2_Surprising.Vihelmo-Jarv.zip)

f8lerror, 2023-08-14:
That worked, Thank you very much

closed 2023-08-14 by f8lerror