Ultimaker/Cura #16473: Error slicing

LoKiki56, 2023-08-13:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic I3 Mega

### Describe your model

Slice OK with 1 model, not with many

### Add your .zip here ⬇️

[PLA Love Zelda 25mm x13.zip](https://github.com/Ultimaker/Cura/files/12329350/PLA.Love.Zelda.25mm.x13.zip)

GregValiant, 2023-08-13:
Thanks for the report and the file.
I can duplicate the problem.
This might be a new bug.  The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic