Ultimaker/Cura #16475: wipe changes direction unexpectedly

TheNecronomica, 2023-08-14:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Pro

### Reproduction steps

Outer wall wipe distance changes directions based on number of walls and the setting wall ordering.
So if the outer wall printed has a inner wall next to it it goes in the correct direction independent of wall ordering setting.
if the outer wall has another outer wall next to it one of these walls will have the wipe direction reversed. (see images below)
Outside to inside:
[image]
Inside to outside:
[image]

Changing the setting "wall ordering" will change if the "inner" or "outer" outside wall goes in the wrong direction (the outer wall connected to an inner wall still goes the correct direction).
This creates an issue where either the "inner" or "outer" outside wall will have a worse z seam instead of both getting better.

In this model in particular this creates problems with tolerances since the "blades" are moving parts and if the z seam is not good the parts will get stuck or become sluggish in movement. so with this bug present i cannot currently print this model. 

### Actual results

Wipe direction changes to go backwards.

### Expected results

Wipe direction to continue in the same direction already traveled regardless of other settings and circumstances. 

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/12333004/cura.log.zip)
[COLLAPSING-KATANA Project file.zip](https://github.com/Ultimaker/Cura/files/12333011/COLLAPSING-KATANA.Project.file.zip)

GregValiant, 2023-08-14:
Thanks for the report and the file.
It's an interesting model and it presents challenges.
The Outer Wall Print Direction is being determined not by whether there is an inner wall next to it, but rather by whether it is indeed an "Outer Wall".  An Outer Wall has air on one side and model on the other so in fact all the outer walls of those separate interior models are extruded in the same direction...air on the left and model on the right.  The fact that the cross section is thin enough (0.92mm) that there are no other wall loops never comes into it and hence the "wipe direction" doesn't have anything to do with it.
If you slice with a 0.20 nozzle so those inside models get inner walls you will see what happens.

For this model I would use the Mesh Tools plugin and "Split model into parts".  I would change the Z seam location of each part using the "Per Model" settings.  If you alternate the Z seam from "Left" to "Back" to "Right" to "Front" for each part then they won't line up to create interferences.
There is only a 0.46mm gap between the sections.  Tolerances are critical on this model and that would be especially true if the Z seams lined up.

After splitting the model into separate parts you could also print the inside blade sections with the Z seam set to "Random" and using the Per Model settings have the handle Z seam in-line.  You can set the Outer Wall Wipe Distance back to something normal like 0.1mm rather than 1mm.
[image]

I'll leave the bug label on this and the Cura team will take a look.
The "Wipe Direction" does not appear to have anything to do with this and the problem seems to be a combination of your settings and "expected behavior" rather than a bug.  This is a close tolerance model and they are never easy.
You could also try using a bit of negative Horizontal Expansion or a bit of Outer Wall Inset to enlarge the gaps between the parts.  You should also set your "Initial Layer Horizontal Expansion" to zero since it is creating a single width wall on the initial layer and bed adhesion is going to be a definite problem.  Since it's tall and thin and you have a bed-slinger printer you might consider inserting an M201 Y350 at about layer 500.  Those inside parts could be shaken loose from the bed pretty easily.

TheNecronomica, 2023-08-14:
That is an interesting solution that I'll have to look into.
I would still call this a bug since after som experimentation it does the same thing on more detailed models that have thinner parts.
the thicker parts that have an inner wall does the wipe in the expected direction and on thinner parts (like horns on a moose or fingers on a small hand) without inner walls the wipe goes "backwards" making the z-seam blob bigger instead of smaller.
i don't have time at the moment but I'll gladly slice another model that is a single piece to show when this problem occurs.
the increased wipe distance was just a way to highlight and make it easier to se what i was pointing to. I don't actually have it set that high.

TheNecronomica, 2023-08-15:
Hi again, attached here is another model and project file to showcase the bug. its seen at the top layer (and perhaps other places, haven't looked further). First two places prints normally with the wipe happening as a continuation of the extrusion move and in the same direction.
The last line has the bug where the wipe happens backwards in the direction the line came from making the z-seam worse instead of better for that specific line (making smaller details look a lot worse since the proportion between wipe and line are closer).
[sigmund-freud-bust-BUG.zip](https://github.com/Ultimaker/Cura/files/12343301/sigmund-freud-bust-BUG.zip)

Screengrabs:
First and second line print directions:
[image]

First and second line wipe direction is correct:
[image]

Last line direction:
[image]

Last line wipe direction going back on itself:
[image]

FountainGoat, 2023-09-25:
Hi

"Alternate wall direction" will also cause this behavior on every other wall when slicing a cylinder with otherwise standard settings + outer wall wipe distance